LennartHennigs/Button2 #4: Button click unreliable

xnaron, 2019-08-10:
I'm finding the single click function doesn't always register (works 80% or less of the time). I'm using a nodemcu V2 and clickable encoder.  I have tried replacing the encoder.  I can see the blue led light up on the nodemcu everytime I click it even when it doesn't register. Perhaps a timing issue?  The sketch is basically built off the demo.  I'm using visual studio code with platformio.  Using the latest encoder library I pulled from github and the button2 library is from platformio.  I tried using your latest one from github and the buttons wouldn't register at all.  

#define ROTARY_PIN1	D2
#define ROTARY_PIN2	D3
#define BUTTON_PIN	D4

/////////////////////////////////////////////////////////////////

ESPRotary r = ESPRotary(ROTARY_PIN1, ROTARY_PIN2, 2 );
Button2 b = Button2(BUTTON_PIN);

void setup() {
  Serial.begin(9600);
    delay(1000);
    Serial.println("\n\nFocusrite Knob");
  
  setupWifi();
   delay(3000);
  broadcast();
  getPort();
  
  setupClient();

  delay(50);

  
  Serial.println("\n\nInitialized");
  
  r.setChangedHandler(rotate);
  r.setLeftRotationHandler(showDirection);
  r.setRightRotationHandler(showDirection);

  b.setClickHandler(muteVol);
  b.setLongClickHandler(resetPosition);

}

void loop() {
  r.loop();
  b.loop();
 }

LennartHennigs, 2019-08-10:
Hm, please try playing around with the denounce time. During this time clicks will be ignored. There is a function to change it. See what happens if you shorten it.

xnaron, 2019-08-10:
Thanks.  I just tried that and it didn't help (1,5,10ms) .  Sometimes it doesn't respond at all.  The rotary encoder is rock solid.  I can see the led on the nodemcu light when I press it but I don't get teh "Button Press" in the serial console.

 b.setClickHandler(muteVol);

void muteVol(Button2& btn) {
Serial.println("Button Press"); 
}

LennartHennigs, 2019-08-11:
Could you try setting the  pressed and the released handlers and see if they also skip?

closed 2020-03-26 by LennartHennigs