LennartHennigs/Button2 #5: SerialBT issue

daveparisee, 2020-03-05:
Hi!
I'm working on esp32 and I need SerialBT lib AND button2.. button2 work great if I comment out SerialBT command in sketch.
As soon as I uncomment SerialBT command, button2 refuse to work. SerialBT is working.

There's no delay is code.

krasniykrd, 2020-03-07:
If u working with esp32 use RTOS tasks for non blocking actions!

LennartHennigs, 2020-03-15:
I'll take some time for me to reproduce this. I'll have to dig up an ESP32 and give it a try.
I usually use ESP8266s so I don't have one at hand.

LennartHennigs, 2020-03-15:
Sorry, I don't  really understand your request. 
In the constructor you can define the button mode  (INPUT or INPUT_PULLUP).
Does this not solve your problem?
If not, could you please elaborate a bit? thx.

daveparisee, 2020-03-15:
Buttons2 and serialBT does not work together..

krasniykrd, 2020-03-16:
> Buttons2 and serialBT does not work together..

On esp32 all works....

LennartHennigs, 2020-03-26:
It might have been, that the loop got stuck because of no delays(). Added one to the `Button2` loop.

closed 2020-03-26 by LennartHennigs