LennartHennigs/Button2 #8: No active HIGH button

Nagymadar, 2020-03-15:
ESP32 can wake up from sleep only for active HIGH, therefore most designs use this. Would be nice if this nice piece of code worked for active HIGH buttons.

Nagymadar, 2020-03-15:
My version to solve this: if you have a negative attachTo input, then it is an active HIGH button. This demands a float input, what distinguishes between -0 (negative zero) and +0 (positive zero). Not the nicest workaround, but 100% backward compatible and works. Other way: using the buttonMode with an enum?

```
Button2::Button2(float attachTo, byte buttonMode /*= INPUT_PULLUP*/, unsigned int debounceTimeout /*= DEBOUNCE_MS*/) {
  if (std::signbit(attachTo)) {
    state_active = HIGH;
	state = LOW;
	pin = -attachTo;
  } else {
    state_active = LOW;
	state = HIGH;
	pin = attachTo;
  }  
  
  setDebounceTime(debounceTimeout);
  pinMode(pin, buttonMode);
}
```

LennartHennigs, 2020-03-15:
Sorry, I don't really understand your request.
In the constructor you can define the button mode (INPUT or INPUT_PULLUP).
Does this not solve your problem?
If not, could you please elaborate a bit? thx.

Nagymadar, 2020-03-25:
The library does not work with active HIGH buttons, only active LOW. If it is active HIGH, it generates a unintentional click_cb call after the object is created, beside of the fact that isPressed is inverted, etc, etc. I modified your library (renamed to Button3), and it works like a charm. Have a look:

button3.h
```
/////////////////////////////////////////////////////////////////
/*
  Button3.cpp - Arduino Library to simplify working with buttons.
  Created by Lennart Hennigs, October 28, 2017.
  Modified by Nagymadar, 2020 March
*/
/////////////////////////////////////////////////////////////////
#pragma once

#ifndef Button3_h
#define Button3_h

/////////////////////////////////////////////////////////////////

#include "Arduino.h"

/////////////////////////////////////////////////////////////////

#define DEBOUNCE_MS      50
#define LONGCLICK_MS    200
#define DOUBLECLICK_MS  300

#define SINGLE_CLICK      1
#define DOUBLE_CLICK      2
#define TRIPLE_CLICK      3
#define LONG_CLICK        4

/////////////////////////////////////////////////////////////////

class Button3 {
  private:
    byte pin;
	int state_active;
    int prev_state;
    int state;
    byte click_count = 0;
    unsigned int last_click_type = 0;
    unsigned long click_ms;
    unsigned long down_ms;
    unsigned int debounce_time_ms;
    unsigned int down_time_ms = 0;
    bool pressed_triggered = false;
    bool longclick_detected = false;
        
    typedef void (*CallbackFunction) (Button3&);

    CallbackFunction pressed_cb = NULL;
    CallbackFunction released_cb = NULL;
    CallbackFunction change_cb = NULL;
    CallbackFunction tap_cb = NULL;
    CallbackFunction click_cb = NULL;
    CallbackFunction long_cb = NULL;
    CallbackFunction double_cb = NULL;
    CallbackFunction triple_cb = NULL;
    
  public:
    Button3(float attachTo, byte buttonMode = INPUT_PULLUP, unsigned int debounceTimeout = DEBOUNCE_MS);
	void clear();
    void setDebounceTime(unsigned int ms);
    
    void setChangedHandler(CallbackFunction f);
    void setPressedHandler(CallbackFunction f);
    void setReleasedHandler(CallbackFunction f);
    void setClickHandler(CallbackFunction f);
    void setTapHandler(CallbackFunction f);
    void setLongClickHandler(CallbackFunction f);
[... 193 lines cut ...]
      }
    }
    click_count = 0;
    click_ms = 0;
  }   
}

void Button3::clear() {
  state = !state_active;
  click_count = 0;
  last_click_type = 0;
  down_time_ms = 0;
  pressed_triggered = false;
  longclick_detected = false;
	
  pressed_cb = NULL;
  released_cb = NULL;
  change_cb = NULL;
  tap_cb = NULL;
  click_cb = NULL;
  long_cb = NULL;
  double_cb = NULL;
  triple_cb = NULL;
}
```

LennartHennigs, 2020-03-26:
Hey  Nagymadar,
thanks for your input. I am currently updating the class and pushing commits to close the issues you posted. (The commits are also linked the issues). 
I'll now test the class a bit and take a look at the suggestion of yours:

```
 if (std::signbit(attachTo)) {
    state_active = HIGH;
	state = LOW;
	pin = -attachTo;
  } else {
    state_active = LOW;
	state = HIGH;
	pin = attachTo;
  }  
```
Best
l.

LennartHennigs, 2020-03-26:
I didn't want to use floats to assign pins to a button. Instead I added a new boolean parameter to the constructor of the Button class: 

`Button2(byte attachTo, byte buttonMode = INPUT_PULLUP, boolean activeLow = true, unsigned int debounceTimeout = DEBOUNCE_MS);
`    

After setting the second parameter to false the class works correctly with active high buttons.
Thank you for your input, Nagymadar.

closed 2020-03-26 by LennartHennigs

Fishbone69, 2023-09-09:
Hi...I tried this lib with active high buttons and still no luck.  Maybe it is my constructor syntax?  I am using the following:
  `buttonUp.begin(BUTTON_UP,INPUT,false); `
although I also tried
 `buttonUp.begin(BUTTON_UP,INPUT_PULLDOWN,false); `
  
Any help is appreciated!