LennartHennigs/Button2 #11: capacitive buttons

qubolino, 2020-06-26:
proposed patch (tested on esp32)

`diff --git a/src/Button2.cpp b/src/Button2.cpp
index 63667b5..4f09d77 100644
--- a/src/Button2.cpp
+++ b/src/Button2.cpp
@@ -10,10 +10,15 @@
 
 /////////////////////////////////////////////////////////////////
 
-Button2::Button2(byte attachTo, byte buttonMode /* = INPUT_PULLUP */, boolean activeLow /* = true */, unsigned int debounceTimeout /* = DEBOUNCE_MS */) {
+Button2::Button2(byte attachTo, boolean isCapacitive /* = false */, byte buttonMode /* = INPUT_PULLUP */, 
+                boolean activeLow /* = true */, unsigned int debounceTimeout /* = DEBOUNCE_MS */) {
   pin = attachTo;
   setDebounceTime(debounceTimeout);
-  pinMode(attachTo, buttonMode);
+  if (!isCapacitive)
+    pinMode(attachTo, buttonMode);
+  else
+    capacitive = true;
+    
   pressed = activeLow ? HIGH : LOW;
   released = activeLow ? LOW : HIGH;
   state = pressed;
@@ -107,7 +112,15 @@ unsigned int Button2::getClickType() {
 
 void Button2::loop() {
   prev_state = state;
-  state = digitalRead(pin);
+  if (!capacitive)
+    state = digitalRead(pin);
+  else{
+    int capa = touchRead(pin);
+    state = capa > 70 ? HIGH : capa < 30 ? LOW : state;
+  }
 
   // is button pressed?
   if (prev_state == pressed && state == released) {`

`diff --git a/src/Button2.h b/src/Button2.h
index 503f2b1..2729aca 100644
--- a/src/Button2.h
+++ b/src/Button2.h
@@ -35,6 +35,7 @@
 class Button2 {
   protected:
     byte pin;
+    bool capacitive = false;
     int prev_state;
     int state;
     int pressed;
@@ -60,7 +61,7 @@ class Button2 {
     CallbackFunction triple_cb = NULL;
     
   public:
-    Button2(byte attachTo, byte buttonMode = INPUT_PULLUP, boolean activeLow = true, unsigned int debounceTimeout = DEBOUNCE_MS);
+    Button2(byte attachTo, bool isCapacitive = false, byte buttonMode = INPUT_PULLUP, boolean activeLow = true, unsigned int debounceTimeout = DEBOUNCE_MS);
     void setDebounceTime(unsigned int ms);
     void reset();`

LennartHennigs, 2020-10-24:
Hey, thanks for the suggested patch. 
I will try it out a ESP and add it if it works!
Cheers.

LennartHennigs, 2020-11-06:
Hi, 
I tested it on a ESP32 FM DevKIt and it works with some minor tweaks.
I published  a modified version to the code. 
I move the parameter after the `INPUT_PULLUP` as this is a more commonly used one. And I added a check for the target architecture, because non ESP32 boards don't "know" the `touchRead()` function.
Best
l.

qubolino, 2020-11-09:
ty

grayconstruct, 2022-02-21:
Would like to see this handled more generically as other devices have a different function than the touchRead()

LennartHennigs, 2022-04-02:
Idea suggested in #32