LennartHennigs/Button2 #12: Two or more keys pressed at the same time.

spairo1741, 2020-07-05:
The M5Stack has three keys A, B, C. How to decrypt the key press twice AB, BC, AC

LennartHennigs, 2020-07-05:
Is the question how would to track simultaneous clicks for buttons? And then, how to track patterns?

I’d record the time of the last click of each button. On a click, I’d the check the previous ones (within a certain timeframe) to see whether a two-button click occurred. 

But when is a two- button click “done”? After two key presses? Or after a time of X milliseconds?
It kind of depends on your definition on how to detect and track them. 

Best 
l.

rabelog, 2021-04-29:
Hello!

@LennartHennigs ,

I have the same issue as spairo1741, 

In my code i'm trying to track 2 buttons,  holding for x milliseconds, after that i do the command when both buttons is released (independent the order, nothing happens if any still pressed).

Please, would you give a council how i can use your library in this case?

Today i use something like that:

```
int test_time(long lastDebounceTime) {
  int aux_time = 0;
  while ((digitalRead(button1) == LOW && digitalRead(button2) == LOW)) {
    if ((millis() - lastDebounceTime) > layoutDelay) {
      if (aux_time == 0) {  // used this aux to do once and then nothing in the "while" after the hold time 
        returnStatusBattery();
        aux_time = 2; // one kind of return
      }
    }
  }

  while ((digitalRead(button1) == LOW && digitalRead(button3) == LOW)) {
    if ((millis() - lastDebounceTime) > layoutDelay) {
      aux_time = 1;
      if (aux_layout == 0) { // used this aux to do once and then nothing in the "while" after the hold time 
        vibraSwitchLayout();
        aux_layout = 1;
      }
    } else {
      aux_time = 0;
    }
  }
  return aux_time; // In this return i know if the button1+2 or button 1+3 was pressed for x milliseconds
}

```

LennartHennigs, 2021-04-29:
I‘ll post an example in a couple of days...

closed 2021-08-30 by LennartHennigs