LennartHennigs/Button2 #14: LongpressHandler demo doesn't work

Anton-V-K, 2020-10-13:
I've borrowed the code from LongpressHandler demo, and it doesn't work.
Then I just tried the demo, and... it doesn't work either.

Platform: ESP8266 (NodeMCU V3) / Arduino IDE 1.8.13 / ESP8266 Core 2.7.4

LennartHennigs, 2020-11-06:
Hi,
tested it (again) on a **WeMos D1 Mini** and it works:

Longpress is defined as 200ms, after that I get the following output:
```
Longpress Handler Demo
You clicked a really really long time. (3988 ms)
You clicked a really really long time. (1614 ms)
You clicked a really long time. (1052 ms)
You clicked a really long time. (1326 ms)
You clicked a really really long time. (1982 ms)
You clicked a long time. (877 ms)
You clicked a long time. (637 ms)
You clicked long. (399 ms)
You clicked long. (327 ms)
You clicked a long time. (522 ms)
You clicked long. (339 ms)
You clicked long. (311 ms)
You clicked long. (213 ms)

```

The wiring was: D7 --> BTN --> GND

Also had similar results with **Adafruit Feather Huzzah ESP8266**. There I used pin 2.
Unfortunately, I cannot reproduce your result.

closed 2020-11-06 by LennartHennigs

waihankan, 2021-01-04:
> I've borrowed the code from LongpressHandler demo, and it doesn't work.
> Then I just tried the demo, and... it doesn't work either - only the 'welcome message' is shown, and no reaction on clicks.
> 
> The 'button' is connected to D1 and GND, so I've adjusted the demo sketch:
> 
> ```
> #define BUTTON_A_PIN  D1
> ```
> 
> Platform: ESP8266 (NodeMCU V3) / Arduino IDE 1.8.13 / ESP8266 Core 2.7.4

Have you made this work? I'm facing a similar problem. In my case, not only the longpresshandler demo but also other demos are not working. Just see the welcome message and nothing's happening in my Serial Monitor