LennartHennigs/Button2 #16: pressed and released reversed in code?

TommyC81, 2020-11-02:
https://github.com/LennartHennigs/Button2/blob/8ab31adacf95f4db3d5410d87adda78794c55f78/src/Button2.cpp#L17

This code:
```
pressed = activeLow ? HIGH : LOW;
released = activeLow ? LOW : HIGH;
```
If activeLow is true (i.e. INPUT_PULLUP pin mode used), shouldn't the (button) 'pressed' state be a LOW reading on the pin? And vice versa of the 'released' state (i.e. 'released' is when you're NOT pressing/pushing the button).

I'm trying to find a decent Button library, but want to understand the source code before I dig in. Just checking if I'm reading the source code correctly. The seemingly reversed references to 'pressed' and 'released' is confusing, chances are that I simply got it wrong. Appreciate any clarification you have.

LennartHennigs, 2020-11-02:
Well, you could just try out the example and see what happens, no?

But to answer your question: the active high refers to [this issue](https://github.com/LennartHennigs/Button2/issues/8). There you'll find the details to _activeLow_ setting.

TommyC81, 2020-11-03:
Thanks Lennart,

the library seems to work fine so far.

It's only when digging through the source that things gets a bit confusing.
It seems that the use of 'released' and 'pressed' variables are reversed in
the code, but it works as the HIGH and LOW pin settings are also configured
in  reverse. Chances are that I'm missing something. If time permits I'll
provide something more tangible, appreciate your time and effort in
availing the library, and responding.

Best regards

TommyC81, 2020-11-04:
See #17  that fixes the reversed usage of variables (and one comment) without affecting the functionality (as far as I can tell). Now the code and comments make much more sense... :-) Hope it explains and helps!

TommyC81, 2020-11-06:
Hi Lennart,

I've added even more fixes and also fixed a potential bug. Have a look at the pull request #17 and consider merging. Change of return data type (byte instead of unsigned int) could perhaps be a problem for some, but I'm not sure.

LennartHennigs, 2020-11-06:
Thx Tommy, 
for the review and the changes to the code. 
Merged it back into the main code!

closed 2020-11-06 by LennartHennigs