LennartHennigs/Button2 #20: Does calling the loop() member function in my sketch's loop() function have any effect on timing function?

waihankan, 2021-01-04:
Hi, Greetings,
Thanks for this amazing library Lennart. I'm planning to use this in my code, which includes an OLED display and a rotary encoder. So, I'm wondering if this library is compatible with my code. With respect!

waihankan, 2021-01-04:
Hi, 
I'm currently trying the longpressHandler example, using pin 8 on Arduino Uno as the button pin. In my case, only "Longpress Handler Demo" is on the screen.

I added `Serial.println("test")` in `void longpress(Button2& btn)` function just to make sure that **"button.loop()"** is just working fine. Unfortunately, I think this part of the code, `void loop() {
  button.loop();
}` is not working currently.

Any clarifications on this issue will be appreciated.
Thanks again.

LennartHennigs, 2021-01-04:
Hey, 
thanks for pointing this out. 
Please check out the latest version I just released. There was a bug I introduced in 1.5.0. 
It should now work fine.
Best
l.

waihankan, 2021-01-04:
:pushpin: Just tested out! Everything works fine. Thanks, Lennart.
:wink: I am grateful for accepting the bug suggestion!

Note: You should release the latest version in Arduino Library Manager too. The beginners might have found problems installing the Arduino library in GitHub. Anyway, great code!

closed 2021-01-04 by waihankan

LennartHennigs, 2021-01-04:
(The Arduino Library Manager catches up on the release by itself – it just takes a few hours)