LennartHennigs/Button2 #21: Partial Reverse

zenturacp, 2021-01-25:
It seems like the isPressed() method is somehow reversed..

When all is behaving as it should - click working etc.. the status of isPressed is 0 when the button is pressed and 1 when the button is released?

Is that by intension or is it a bug?

I have this code running on a Wemos ESP8266 Mini D1 Lite what what is the reference for isPressed - imo it should return 1 if the button is pressed, and 0 if released

I have tried to init the button with activeLow = false also, but still i always endup getting the "reverse" of what i expect in the isPressed() method..

```
#include <Arduino.h>
#include <Blink.h>

// #define DEBOUNCE_MS 100
#include <Button2.h>

// int LED_PIN = D8;
int LED_PIN = LED_BUILTIN;
int BTN_PIN = D1;

unsigned long lastUpdate = millis();

Button2 btn(BTN_PIN);

void statusUpdate()
{
  if ((millis() - lastUpdate) > 1000)
  {
    bool btnPressed = btn.isPressed();
    lastUpdate = millis();
    if (btnPressed == 0)
    {
      Serial.println("Pressed");
      Serial.print("Library: ");
      Serial.println(btnPressed);
      Serial.print("Digital Read: ");
      Serial.println(digitalRead(BTN_PIN));
    }
    else
    {
      Serial.println("Released");
      Serial.print("Library: ");
      Serial.println(btnPressed);
      Serial.print("Digital Read: ");
      Serial.println(digitalRead(BTN_PIN));
    }
  }
}

void btnClick(Button2& btn) {
    Serial.println("Click\n");
}

void btnTap(Button2 &btn)
{
  Serial.println("Tap\n");
  digitalWrite(LED_PIN, LOW);
  delay(100);
  digitalWrite(LED_PIN, HIGH);
  delay(100);
}

void setup()
{
  // put your setup code here, to run once:
  Serial.begin(9600);
  delay(100);
  Serial.println("Startup!");
  delay(50);
  pinMode(LED_PIN, OUTPUT);
  digitalWrite(LED_PIN, HIGH);
  delay(1000);
  btn.setTapHandler(btnTap);
  btn.setClickHandler(btnClick);
}

void loop()
{
  btn.loop();
  statusUpdate();
}
```

LennartHennigs, 2021-01-26:
Hey Christian,   
thanks for reporting  this. Yes, you are right.
This is a bug. I updated the source and will release an update.

I used your example and modified it a bit to test.
Not sure, if you find it helpful.
If not, ignore...
Cheers
l.

```
#include "Button2.h"

int LED_PIN = LED_BUILTIN;
int BTN_PIN = D3;

unsigned long lastUpdate = millis();
Button2 btn;

void statusUpdate() {
  if ((millis() - lastUpdate) > 100) {
    Serial.print(btn.isPressed());
    lastUpdate = millis();
  }
}

void setup() {
  Serial.begin(9600);
  delay(100);
  Serial.println("Startup!");
  delay(50);

  pinMode(LED_PIN, OUTPUT);
  digitalWrite(LED_PIN, LOW);

  btn = Button2(BTN_PIN);
  btn.setClickHandler([](Button2& btn) {
    Serial.println(" --> Click");
  });
  btn.setPressedHandler([](Button2 &btn){
    Serial.println();
    digitalWrite(LED_PIN, HIGH);
  });
  btn.setReleasedHandler([](Button2 &btn){
    Serial.println();
    digitalWrite(LED_PIN, LOW);
  });
}

void loop() {
  btn.loop();
  statusUpdate();
}

```

zenturacp, 2021-01-26:
> Hey Christian,
>
> thanks for reporting  this. Yes, you are right.
>
> This is a bug. I updated the source and will release an update.
>
>
>
> I used your example and modified it a bit to test.
>
> Not sure, [...]

THANK YOU - I really was trouble shooting all my inputs and ever until I read the actual code 🤣 i was really getting crazy - was just using the method for debugging but it was saying the opposite as it was supposed to do..

Awesome that you have fixed it - really good work with this very tiny library

closed 2021-01-26 by LennartHennigs