LennartHennigs/Button2 #23: Button2 Class has no member getAttachPin

madivad, 2021-02-08:
I am referencing a sketch that uses Button2 and I was pointed to this repo, however there is an error I am getting about `getAttachPin`, in searching your code I see that it in fact doesn't appear. 

I have found several sketches that reference this non-existant class member:

- I am using a sketch located here: https://github.com/Xinyuan-LilyGO/LilyGo_Txx
- but others exist here: https://github.com/LilyGO/TTGO-T-ControllerV2.2/blob/master/T18_V2.2/adc.ino
- here: https://github.com/Xinyuan-LilyGO/T5-Ink-Screen-Series/blob/master/src/main.cpp
- and here: https://www.tedizen.com/ttgo-tbadge-getting-started-tutorial/

I am using your current master as of this date (1.5.3)

I am not sure where the error is, I can't see any member that returns a pin from the class. Is this something that other parties have created as an extension to your class? It's funny that several sketches refer to it

LennartHennigs, 2021-02-08:
Hey, 
you are correct in both instances – yes, there is no function like this in my class and yes, I did see a fork where [lewisxhe](https://github.com/lewisxhe/Button2) implemented it. 

Well, I never thought it was really needed, since you assign the pin in the constructor or in the `begin()` function and you most likely have the pin defined  it as a constant or via a `#define` on the top of you sketch. 

But I am could be convinced otherwise :-)  Any thoughts on why it makes sense to have it? 

Cheers
l.

madivad, 2021-02-08:
I only just found the fork in the last few minutes, I'm working through some example code and I haven't got into the hows and whys yet. 

I think in the cases I have seen it's been the fact that various buttons are mapped and they would want to be determining which button has activated the callback. 

I was trying to rework the original code because I could see that's what they were trying to achieve, but this level of .cpp and callbacks and references is a little beyond my level of understanding.

LennartHennigs, 2021-02-08:
Well, its a quick fix. I'll add it to the main branch and release it today.
Will close this when I push it. 
Do you download the source from here or shall I make it a proper version and push it Arduino IDE/Platform.io as well?

madivad, 2021-02-08:
I did download it through the Arduino IDE, I just edited the library file directly, so hopefully when the new version is pushed my manual edit will be replaced by yours, I'll check it myself later. Thanks for your quick and speedy response to this, it did have me pulling my hair out since my google fu was not finding the fork initially. I never even thought to check for that. Thanks again

LennartHennigs, 2021-02-08:
created a new version and it wil be available through the Arduino IDE or here via download.

closed 2021-02-08 by LennartHennigs