LennartHennigs/Button2 #25: Detect long press while button is pressed

violetforest, 2021-03-24:
Hi, from what I understand in the code now, the long press only triggers the callback when the long press is finished. Is there a way to detect a long press while the button is still being pressed?

LennartHennigs, 2021-03-24:
Yes, it is. 

- The callback function from setLongClickDetectedHandler() will be called as soon as the long click timeout has passed.

- The callback for setLongClickHandler() will be called after the button has released.

violetforest, 2021-03-24:
Ah I see, but I suppose I wouldn't be able to get that handler in a loop, with a looping log of the time that it is being pressed down for?

LennartHennigs, 2021-03-24:
The  core concept of this Library is getting feedback via handlers. You seem to be looking for something different. 

But you could use isPressed() or the setPressedHandler() and store the timestamp and work with this in your core loop.

closed 2021-03-24 by violetforest