LennartHennigs/Button2 #28: How to handle initial button state?

giannoug, 2021-08-04:
Consider the following button:
`Button2 bigRedButton = Button2(bigRedButtonPin, INPUT_PULLUP, false, false);`

For some weird reason, the button is pressed while the Arduino reboots. This doesn't fire any callbacks. I have tried `setPressedHandler`, `setReleasedHandler` and `setChangedHandler`. Since the button is not active low, shouldn't it trigger on boot? It works fine if I release and press it after booting.

LennartHennigs, 2021-08-30:
Hey George,  I will try to replicate this issue.

LennartHennigs, 2021-08-31:
Hey George, I have a hard time duplicating the behavior you describe. 
But please checkout the latest version via GitHub and give it a try. I updated how I set the Initial values of the `state` and `prev_state` vars. Maybe this will solve your issue.

closed 2021-09-12 by LennartHennigs