LennartHennigs/Button2 #33: Active low button don't work with the library

henkjannl, 2022-03-13:
Perhaps really silly question, but none of the examples work with my ESP32. I have 3 buttons switching to ground. They work perfectly reliable when just reading the buttons using pinMode(#, INPUT_PULLUP) and digitalRead(#) (0=pressed, 1=not pressed). Very surprisng since the default of the library also seems to be INPUT_PULLUP.
I'm sure it has something to do with configuring the Button2 button using the constructor or begin(). 
Is it possible to document this configuration in the Readme file?
Many thanks.

LennartHennigs, 2022-03-13:
Hey, 
I have been using my lib with ESP32s. 
What ESP are using?
And did you change the example code (other than the pins)?

Cheers
l.

henkjannl, 2022-03-13:
Thanks for the quick response Lennart. 
I'm using a MH-ET Live Minikit with ESP-WROOM-32 like this:

[image]

In reality, a ILI9341 and stepper motor are connected to other pins, but I don't think they cause the problem.

I used this code to prove the electrical connections are working:
```c++
int BTN1 = 21;
int BTN2 = 22;
int BTN3 = 15;

// the setup routine runs once when you press reset:
void setup() {
  Serial.begin(115200);
  pinMode(BTN1, INPUT_PULLUP);
  pinMode(BTN2, INPUT_PULLUP);
  pinMode(BTN3, INPUT_PULLUP);
}

// the loop routine runs over and over again forever:
void loop() {
  Serial.print(digitalRead(BTN1));
  Serial.print(digitalRead(BTN2));
  Serial.println(digitalRead(BTN3));
  delay(100);    
}
```

This does not provide a response on the handler:
```c++
#include "Button2.h"

#define BUTTON_PIN_1  21
#define BUTTON_PIN_2  22
#define BUTTON_PIN_3  15

Button2 button_1, button_2, button_3;

void setup() {
  Serial.begin(115200);
  delay(50);
  Serial.println("\n\nMulti Handler Demo w/ 2 buttons");

  button_1.begin(BUTTON_PIN_1);
  button_1.setClickHandler(handler);
  button_1.setDoubleClickHandler(handler);

  //button_2.begin(BUTTON_PIN_2);
  button_2.setClickHandler(handler);
  button_2.setDoubleClickHandler(handler);

  //button_3.begin(BUTTON_PIN_3);
  button_3.setClickHandler(handler);
  button_3.setDoubleClickHandler(handler);
}

void loop() {
  button_1.loop();
  button_2.loop();
  button_3.loop();

  Serial.print( (button_1.isPressed()) ? "X" : "-");
  Serial.print( (button_2.isPressed()) ? "X" : "-");
  Serial.print( (button_3.isPressed()) ? "X" : "-");
  Serial.println();
  
  delay(200);
}

void handler(Button2& btn) {
    switch (btn.getClickType()) {
        case SINGLE_CLICK:
            Serial.print("single ");
            break;
        case DOUBLE_CLICK:
            Serial.print("double ");
            break;
    }
    Serial.print("click ");
    Serial.print("on button #");
    Serial.print((btn == button_1) ? " 1 " : " - ");
    Serial.print((btn == button_2) ? " 2 " : " - ");
    Serial.print((btn == button_3) ? " 3 " : " - ");
    Serial.println();
}
```

Also, if I comment out:

```c++
  Serial.print( (button_1.isPressed()) ? "X" : "-");
  Serial.print( (button_2.isPressed()) ? "X" : "-");
  Serial.print( (button_3.isPressed()) ? "X" : "-");
  Serial.println();
  
  delay(200);
```
it is still not working.

I tried fiddling with the constructor and begin() but I did not systematically record what I did.

HenkJan

LennartHennigs, 2022-03-14:
Hey, 
I looked at your on a *M5Stack Core ESP32*.
It worked for me out of the box after I deleted  the `delay()` and the `printlns()` in the `loop()`:

```
#include "Button2.h"

#define BUTTON_PIN_1  39
#define BUTTON_PIN_2  38
#define BUTTON_PIN_3  37

Button2 button_1, button_2, button_3;

void setup() {
  Serial.begin(9600);
  delay(50);
  Serial.println("\n\nMulti Handler Demo w/ 2 buttons");

  button_1.setClickHandler(handler);
  button_1.setDoubleClickHandler(handler);

  button_2.setClickHandler(handler);
  button_2.setDoubleClickHandler(handler);

  button_3.setClickHandler(handler);
  button_3.setDoubleClickHandler(handler);

  button_1.begin(BUTTON_PIN_1);
  button_2.begin(BUTTON_PIN_2);
  button_3.begin(BUTTON_PIN_3);
}

void loop() {
  button_1.loop();
  button_2.loop();
  button_3.loop();
}

void handler(Button2& btn) {
    switch (btn.getClickType()) {
        case SINGLE_CLICK:
            Serial.print("single ");
            break;
        case DOUBLE_CLICK:
            Serial.print("double ");
            break;
    }
    Serial.print("click ");
    Serial.print("on button #");
    Serial.print((btn == button_1) ? " 1 " : " - ");
    Serial.print((btn == button_2) ? " 2 " : " - ");
    Serial.print((btn == button_3) ? " 3 " : " - ");
    Serial.println();
}
```

(And I used three button pins of the M5Stack in the code above.)

I briefly looked at the pinout of [your ESP32](https://doc.riot-os.org/group__boards__esp32__mh-et-live-minikit.html#esp32_mh_et_live_minikit_pinout): 22 and 21 are also used for I2C on the board. 
But you checked the pins, as you wrote, right? 

The only problem I could think of, is that you maybe had still a `delay()` inside your `loop()`.
That would break the *button2* `loop()` .

Hope this helps...

closed 2022-03-17 by LennartHennigs

henkjannl, 2022-03-19:
Thanks Lennart, now I understand the loop needs to be called frequently. Presumably my screen update routine takes too long to work well with the library. I still don't get why the out of the box examples did not work though.

LennartHennigs, 2022-12-13:
FYI, you could use an interrupt timer to call the loop. I added an example to the lib.