LennartHennigs/Button2 #38: Decalring SetHandler time

rassalrich, 2022-04-27:
Hi I am trying to use your library for my code, everything seems to be working on compile. However, I am getting an error code saying src\main.cpp: In function 'void setup()':
src\main.cpp:89:27: error: 'click' was not declared in this scope
   buttonA.setClickHandler(click);

I am using the multiple buttons part from your code

LennartHennigs, 2022-04-27:
Hey, did you define a function called click() in your source code? 
This is what the compiler is looking for.

rassalrich, 2022-04-27:
No, I haven't defined the same, since your example code had no part defining the same. How would you advice to have the same defined?

LennartHennigs, 2022-04-27:
Please take another look at the example:

examples/MultipleButtons/MultipleButtons.ino

Below the loop() there IS the click() function defined.

closed 2022-04-27 by LennartHennigs