LennartHennigs/Button2 #42: Read initial value during boot

guydvir2, 2022-10-05:
Hi.
I'm using a 2 state physical switch and an ESP32.
The switch is constant ON, or constant OFF (act as a light switch), and connected as ACTIVE_LOW.

At boot, switch can be "ON" (`LOW`). 
In that case, right after boot sequence, light is switched ON, meaning that `setPressedHandler` handler called.

Firstly, I'm looking for a way that ON-BOOT-STATE of the switch will be saved as start state.
Secondly, from what I see, inside `begin` function that state is supposed to be stored, but still, it is not solved:
```
 state = _getState();
  prev_state = state ; 
```
Thirdly, adding `button2_instance.loop()` right after `button2_instance.begin(Pin)`, actually solved is, but still it looks awkward.

Help any assistance to understand how to store ON-BOOT-STATE, for such cases.

Guy

LennartHennigs, 2022-10-31:
Hey @guydvir2,
well, as you said  – you are using the class for a light switch and not for a button :-)

Upon instantiation, the class detects the state of the switch as pressed.
Thus the pressed handler is called and a click or a long click will be called when the switch status changes.
This sounds like the correct behaviour for the class.
It detects the button a being pressed and reports the release. 

You are now asking to optionally neglect the pressed state upon start, correct?

guydvir2, 2022-11-02:
> Hey @guydvir2, well, as you said – you are using the class for a light switch and not for a button :-)
>
> Upon instantiation, the class detects the state of the switch as pressed. Thus the pressed [...]

Yes :)

ryancasler, 2022-11-04:
How exactly are you going to use this library to handle switches?  Are you only going to look at the tapped and released events?  It seems like an awfully big waste to only use those events.  You may want to look at the [ez-switch-library](https://github.com/ronbentley1/eazy-switch-library) instead.

LennartHennigs, 2022-12-13:
Sorry, I decided that I will not add switch functionality to this lib.

closed 2022-12-13 by LennartHennigs