LennartHennigs/Button2 #44: Disable double/triple click detection

pablopeu, 2022-10-10:
I'w writing a BT BLE keyboard to remotely control a machine via keyboard shortcuts, the most used keys are the arrows, and sometimes you click these a lot to get to the exact position. 
Is there a way to disable repetitive click to be detected as double/triple clicks? 
I can hack this by duplicating the detection and issuing double/triple the amount of keyboard taps on the respective handlers, but I guess a DEFINE would be far more elegant. Same for long press would be the icing in the cake :)

Thanks!

LennartHennigs, 2022-10-30:
Hey @pablopeu,
thanks for the feedback. I added two compliler switches (_IGNORE_DOUBLE and _IGNORE_TRIPLE) to the code.
It is not in the released version yet, as I want to add a longpress switch as well. 
But if you download the latest version you can already use it.
Cheers
l.

pablopeu, 2022-10-30:
Thanks a ton!

LennartHennigs, 2022-10-30:
(Almost done - still have to do long click...) 
:-)
You are welcome.

pablopeu, 2022-10-30:
While you are at it, with the disable longpress switch, will I be able to detect two/three keys pressed at once? Think of CTRL-ALT-DEL or CTRL-X when emulating a keyboard like I'm doing. Thanks

LennartHennigs, 2022-10-30:
I knew you'd ask. 
Take a look at `TrackDualButtonClick`. Set flags for "special" keys and go from there. 
But if you do a real keyboard, I'd look for a keyboard matrix library instead, I think. Less pins to take care of.

pablopeu, 2022-10-30:
You made me grin! Thanks again. Its not a full keyboard, just a few keys for a custom remote keyboard, trying to avoid complexity by not using a keyboard matrix.

LennartHennigs, 2022-10-31:
done.

pablopeu, 2022-10-31:
Hi, added the #define at the top of my program, downloaded latest library code, overwrited previous one, uploaded new code to ESP32 but doubleclicks do not behave like rapid single clicks, made a short video: [https://www.youtube.com/watch?v=b83OWDCFgO4](url)

LennartHennigs, 2022-10-31:
Hey, try to uncomment the defines in the library itself.

pablopeu, 2022-10-31:
Did it and saw the compilation messages about the routines not being compiled, but still doesn't catch a fast click, maybe its because Im reading 12 buttons, dunno.

LennartHennigs, 2022-11-01:
Hey,
I think you don't see some of the "fast clicks" because of the thresholds for double and triple clicks are still there.
A way around this could be to use `setTapHandler()`.
This one is triggered on any click.

pablopeu, 2022-11-01:
will look into it and let you know! Thanks Lennart

pablopeu, 2022-11-03:
Did a quick test that worked, then did a find & replace on my full program: setClickHandler to setTapHandler and recompiled. Man, what a difference in responsiveness, its super fast on all the defined buttons (10 at this moment)

Thanks for the suggestion. BTW, is there a downside to using setTapHandler?

LennartHennigs, 2022-11-03:
No. No downside. It does not respect the double or triple click timings. Thus, it reports clicks immediately.

LennartHennigs, 2022-11-03:
pushed new version (2.1.0)

LennartHennigs, 2022-12-13:
Hey, I decided to remove the compiler switches as the tap handler solves your question.  They only made the code more complex and did not really reduced the compiled code size.   
cheers

pablopeu, 2022-12-13:
you're the boss!!! The tap handler did solve my issue. Thanks for all the help!!

closed 2022-12-18 by LennartHennigs