LennartHennigs/Button2 #45: Error Compiling For ESP32 C3

ryancasler, 2022-11-01:
When compiling a Button2 sketch using an ESP32 C3 board (I tried 2 different brands, DFRobot and Adafruit), I get the following error:
```
c:\LIBRARY DIRECTORY\Button2\src\Button2.cpp: In member function 'byte Button2::_getState()':
c:\LIBRARY DIRECTORY\Button2\src\Button2.cpp:293:18: error: 'touchRead' was not declared in this scope
       int capa = touchRead(pin);
                  ^~~~~~~~~
c:\LIBRARY DIRECTORY\Button2\src\Button2.cpp:293:18: note: suggested alternative: 'timerRead'
       int capa = touchRead(pin);
                  ^~~~~~~~~
                  timerRead
exit status 1
Compilation error: exit status 1
```
One thing that sets the C3 apart from other members of the ESP32 family is the fact that it doesn't support touch.  I can't seem to figure out why it's reading this as capacitive.  Here are the relevant sections of my sketch:

```
Button2 button_0, button_1, button_2, button_3, button_4, button_5, button_6, button_7;
Button2 myButtons[] = { button_0, button_1, button_2, button_3, button_4, button_5, button_6, button_7 };
int buttonPins[] = {A0, A1, A2, A3, SDA, SCL, TX, RX};
int numButtons = 8;

void setup() {
  Serial.begin(115200);
  delay(100);
  Serial.println("Initialized");
  for (int i = 0; i < numButtons; i++) {
    myButtons[i].begin(buttonPins[i], INPUT_PULLUP, false, true);
    myButtons[i].setClickHandler(clickHandler);
    myButtons[i].setDoubleClickHandler(doubleHandler);
    myButtons[i].setLongClickDetectedHandler(longHandler);
    myButtons[i].setReleasedHandler(released);
    myButtons[i].setLongClickTime(800);
    myButtons[i].setDoubleClickTime(400);
    myButtons[i].setDebounceTime(75);
  }
}
```

And of course I have all the relevant callbacks and whatnot.  

Now, If I switch to an ESP32 S2 board, the exact same sketch compiles perfectly with zero changes.  So, I assume there is something wrong in the board definition or the variants pin definition for the C3 boards but for the life of me I can't find it.   I also tried to define a custom state function for all of the buttons and that didn't help.  I still got the same error when compiling.  Any advice on where to look and what might need to be changed?  I'm using the latest version of the ESP32 Core for Arduino, v2.0.5.
https://github.com/espressif/arduino-esp32/releases/tag/2.0.5
Thanks so much!!

ryancasler, 2022-11-01:
I found a temporary workaround.  If I change line 289 of the published version of the Button2.cpp file from this:
`if (!is_capacitive) {`
to this:
`if (!1=1) {`
Then it always compiles correctly.  So, there is definitely something about the definition of those pins that is making it think those pins are capacitive.  And none of the pins on a C3 can be capacitive.

LennartHennigs, 2022-11-01:
Hey @ryancasler,
yes, there is a problem with the C3s. 
Back then I assumed that all ESP32s have capacitive touch. 
In line 313 of `Button2.cpp`I check for this.

Could you try this sketch and tell me if all your C3s are detected?

```
#if defined(ESP32) 
  #pragma message "Using ESP32"
  #if defined(ARDUINO_ESP32C3_DEV) 
    #pragma message "Using ESP32_C3"
  #endif
#elif defined(ESP8266) 
  #pragma message "Using ESP8266"
#else
  #pragma message "Using something else"
#endif

void setup() { }
void loop()  { }
```

LennartHennigs, 2022-11-02:
Hey @ryancasler,
I decided that I don't want to add another special case for the exclusion of the C3.
Instead, using capacitive touch is the special case. 
I will move the cap. touch code out of the library and use the custom handler to provide an example.

LennartHennigs, 2022-11-03:
pushed new version

closed 2022-11-03 by LennartHennigs

ryancasler, 2022-11-07:
Yup...new version worked perfectly.   Thanks!!!