LennartHennigs/Button2 #47: Filtering out Single press while waiting for the Long Press

mrdROb, 2022-11-20:
I've got a button, with setLongClickHandler, and setPressedHandler
When I press button shortly, my PressedHandler works fine. But it also triggers when I do LongPress. So, I get an event just once I pressed the button, and another event in few ms (long press detected)

So the question is - how to filter out the Short Press events when waiting for the LongPress?

mrdROb, 2022-11-20:
Don't know if this is the most appropriate way, but i moved from PressedHandler to Released one. Added a simple check that it only works if press was shorter than Long press.

`btn2.setReleasedHandler([](Button2 & b) {
        if (btn2.wasPressedFor()<LONGCLICK_MS) {
          btnCick = false;
          Serial.println("Single press");
          wifi_scan();
        }
    });`

LennartHennigs, 2022-11-21:
Hey Sergey,
you are now using a general handler (released) and a specific one (longClick).
An alternative would be to use the specific one for the click: `setClickHandler`, or use two handlers.
Cheers
l.

mrdROb, 2022-11-21:
My bad! Thanks Lennart! Looks most appropriate to use setClickHandler, which i did not see :)
Cheers

closed 2022-11-21 by LennartHennigs