LennartHennigs/Button2 #49: Long click retriggerable stopped working in a recent update.

tfabris, 2022-12-17:
I don't know which version had this happen, but something changed in the long click and retriggerable features recently, and now my "held down" buttons don't work, though they worked with an earlier version of the library. Can you check to see if any of your recent updates caused this? Is this a known issue?

Attached is my code. Perhaps you can glean something from it?

[Master Bedroom LEDs.zip](https://github.com/LennartHennigs/Button2/files/10250212/Master.Bedroom.LEDs.zip)

LennartHennigs, 2022-12-17:
Hey, thanks for reaching out. 

Yes, I introduced a bug in 2.2.1. please check the latest version 2.2.2 this should solve it. Cheers.

tfabris, 2022-12-17:
Awesome, thanks so much. I will try it tomorrow when I return home. Timing wise, it looks like you were releasing the bugfixed version right around the time I was trying to diagnose the issue and post the bug. :-)

Super useful library, thanks so much.

LennartHennigs, 2022-12-17:
Thank you. You are welcome.
Please let me know, if it solves the issue. But it should.🤞

tfabris, 2022-12-18:
Yep, version 2.2.2 solved the issue. Thanks!

closed 2022-12-18 by tfabris