LennartHennigs/Button2 #51: detection of long presses does not work properly inside of an interupt timer

EtoTen, 2023-01-06:
FYI, as you predicted when using this library inside an interrupt timer on an esp32, it seems to fail to detect long clicks.  Every click, long or short, irregardless of .setLongClickTime() is detected as a long click. I haven't delved into why this behaviour is being exhibited but was able to create a quick work around by capturing the start and stop time of the button presses / release within my own application.

Here is a snippet:

`
Button2 buttonMode;
hw_timer_t *mButtonTimer = NULL;

void IRAM_ATTR buttonTimer()
{
  // process buttons
  buttonMode.loop();
}

setup() 
{
  mButtonTimer = timerBegin(0, 80, true);
  timerAttachInterrupt(mButtonTimer, &buttonTimer, true);
  timerAlarmWrite(mButtonTimer, 10000, true); // every 0.1 seconds
  timerAlarmEnable(mButtonTimer);

  buttonMode.begin(MODE_BUTTON, INPUT, false);
  buttonMode.setLongClickHandler(longModeClickStart);
  buttonMode.setLongClickDetectedHandler(longModeClickStop);
}

long modeStart;

void longModeClickStart(Button2 &btn)
{
  modeStart = micros();
}

void longModeClickStop(Button2 &btn)
{
  int secondsSinceLast = round((micros() - modeStart) / 1000000);
  Serial.print("Seconds since last mode press: ");
  Serial.println(secondsSinceLast);
  if (secondsSinceLast > 20)
  {
   Serial.println("doing something after 20 seconds");
  }
}
`

LennartHennigs, 2023-01-07:
Hey Nickolai,
thank you for raising the issue! And of course, it doesn't! 
After thinking about it for a minute over coffee: I use `millis()` inside my code. 🤦 
That's at least one reason without looking into the code. 
I will take a look at it in the next couple of days.

LennartHennigs, 2023-06-21:
To everyone trying this out: do this at your own risk!
Using this inside interrupt function **may very likely not work**.
This library was not designed to be used inside interrupts.