LennartHennigs/Button2 #60: 2.3.0 breaks at least on RP2040 with type errors

kilrah, 2024-01-02:
Not happy about pin identifiers and mode...

```In file included from c:\Users\me\Documents\Arduino\libraries\Button2\src/Button2.h:22,
                 from D:\Electronique\PicoPDSupply\PicoPD_LVGL\PicoPD_LVGL.ino:8:
c:\Users\me\Documents\Arduino\libraries\Button2\src/Hardware.h: In member function 'virtual void ArduinoHardware::pinMode(int, int)':
c:\Users\me\Documents\Arduino\libraries\Button2\src/Hardware.h:42:24: error: invalid conversion from 'int' to 'PinMode' [-fpermissive]
   42 |         ::pinMode(pin, mode);
      |                        ^~~~
      |                        |
      |                        int
In file included from c:\users\me\appdata\local\arduino15\packages\rp2040\hardware\rp2040\3.6.2\arduinocore-api\api\Interrupts.h:8,
                 from c:\users\me\appdata\local\arduino15\packages\rp2040\hardware\rp2040\3.6.2\arduinocore-api\api\arduinoapi.h:29,
                 from C:\Users\me\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.6.2\cores\rp2040/api/ArduinoAPI.h:2,
                 from C:\Users\me\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.6.2\cores\rp2040/Arduino.h:28,
                 from C:\TEMP\arduino\sketches\870F04C926EA2EED0B213104A8CED5D4\sketch\PicoPD_LVGL.ino.cpp:1:
c:\users\me\appdata\local\arduino15\packages\rp2040\hardware\rp2040\3.6.2\arduinocore-api\api\Common.h:98:44: note:   initializing argument 2 of 'void pinMode(pin_size_t, PinMode)'
   98 | void pinMode(pin_size_t pinNumber, PinMode pinMode);
      |                                    ~~~~~~~~^~~~~~~
c:\Users\me\Documents\Arduino\libraries\Button2\src/Hardware.h: In member function 'virtual void ArduinoHardware::digitalWrite(int, int)':
c:\Users\me\Documents\Arduino\libraries\Button2\src/Hardware.h:45:29: error: invalid conversion from 'int' to 'PinStatus' [-fpermissive]
   45 |         ::digitalWrite(pin, value);
      |                             ^~~~~
      |                             |
      |                             int
c:\users\me\appdata\local\arduino15\packages\rp2040\hardware\rp2040\3.6.2\arduinocore-api\api\Common.h:99:51: note:   initializing argument 2 of 'void digitalWrite(pin_size_t, PinStatus)'
   99 | void digitalWrite(pin_size_t pinNumber, PinStatus status);
      |                                         ~~~~~~~~~~^~~~~~
```

LennartHennigs, 2024-01-02:
Hey, thanks! And, oh, great...
Does this work?

```
#ifdef ARDUINO_ARCH_RP2040
class ArduinoHardware : public Hardware {
public:
    int digitalRead(int pin) {
        return ::digitalRead(pin);
    }

    void pinMode(int pin, int mode) {
        ::pinMode(pin, static_cast<PinMode>(mode));
    }

    void digitalWrite(int pin, int value) {
        ::digitalWrite(pin, static_cast<PinStatus>(value));
    }
};
#else
class ArduinoHardware : public Hardware {
public:
    int digitalRead(int pin) {
        
        return ::digitalRead(pin);
    }
    void pinMode(int pin, int mode) {
        ::pinMode(pin, mode);
    }
    void digitalWrite(int pin, int value) {
        ::digitalWrite(pin, value);
    }
};
#endif
```

kilrah, 2024-01-02:
Yup that's fine!

LennartHennigs, 2024-01-02:
perfect. Thank you! Will push an update...

closed 2024-01-02 by LennartHennigs