LennartHennigs/Button2 #61: 2.3.x breaks default implementation?

TheBay0r, 2024-01-27:
Hi! 

Sry if the title might be confusing, but so is the issue I experience.

I have reduced my button code to the core to be able to reproduce it. 
This code seems to be working fine on version 2.2.4 of the library, but doesn't recognise my button button anymore from 2.3.0 and 2.3.1 

```cpp
#include <Arduino.h>
#include <Button2.h>

#define BUTTON_PIN 19

Button2 button;

void setup() {
  Serial.begin(115200);
  Serial.println("Starting Setup");

  button.begin(BUTTON_PIN);

}

void loop() {
  button.loop();

  clickType buttonState = button.read();

  if (buttonState != clickType::empty) {
    Serial.println("Button State not empty");
  }
}
```
Because I first thought my button broke I even tried to just short PIN and GND. But I experienced the same behaviour with shorting.

Help or explanation would be highly appreciated.

LennartHennigs, 2024-01-27:
Hey, 
hm, that is strange.

If you [compare the versions](https://github.com/LennartHennigs/Button2/compare/2.2.4...2.3.1) you'll see that I did not change that much in `Button.h` or `Button.cpp`. I do a more thorough reset of the button state in the `read()` function but that looks correct. 
Unfortunately, I don't have time this weekend to look into this further. 
I'll be a few days.

TheBay0r, 2024-01-27:
Yes, I did compare the versions, which confuses me even more. Sadly it is perfectly reproducible for me. Upgrading, downgrading, uninstalling, reinstalling. 2.2.4 works, above doesn't. If there is anything I can add for debugging purposes, let me know

jay2216, 2024-02-17:
2.3 doesn't work for me either on an Arduino r4. I get several errors.

Button2/src/Hardware.h:59:28: error: invalid conversion from 'int' to 'PinMode' [-fpermissive]

packages/arduino/hardware/renesas_uno/1.0.5/cores/arduino/api/Common.h:96:6: note:   initializing argument 2 of 'void pinMode(pin_size_t, PinMode)' void pinMode(pin_size_t pinNumber, PinMode pinMode);

Button2/src/Hardware.h:62:34: error: invalid conversion from 'int' to 'PinStatus' [-fpermissive] ::digitalWrite(pin, value);

.arduino15/packages/arduino/hardware/renesas_uno/1.0.5/cores/arduino/api/Common.h:97:6: note:   initializing argument 2 of 'void digitalWrite(pin_size_t, PinStatus)' void digitalWrite(pin_size_t pinNumber, PinStatus status);

``

LennartHennigs, 2024-02-17:
@jay2216 different issue, create a new issue for it:
#66