LennartHennigs/Button2 #66: Arduino Uno R4 does not work with 2.3

LennartHennigs, 2024-02-17:
2.3 doesn't work for me either on an Arduino r4. I get several errors.

Button2/src/Hardware.h:59:28: error: invalid conversion from 'int' to 'PinMode' [-fpermissive]

packages/arduino/hardware/renesas_uno/1.0.5/cores/arduino/api/Common.h:96:6: note:   initializing argument 2 of 'void pinMode(pin_size_t, PinMode)' void pinMode(pin_size_t pinNumber, PinMode pinMode);

Button2/src/Hardware.h:62:34: error: invalid conversion from 'int' to 'PinStatus' [-fpermissive] ::digitalWrite(pin, value);

.arduino15/packages/arduino/hardware/renesas_uno/1.0.5/cores/arduino/api/Common.h:97:6: note:   initializing argument 2 of 'void digitalWrite(pin_size_t, PinStatus)' void digitalWrite(pin_size_t pinNumber, PinStatus status);

``

_Originally posted by @jay2216 in https://github.com/LennartHennigs/Button2/issues/61#issuecomment-1950021117_

LennartHennigs, 2024-02-17:
Hey, 
thanks for reaching out.
this problem is due to the fact that the R3 uses a Cortex-M4 chip. 
Will take a look

LennartHennigs, 2024-02-17:
Hey @jay2216,
 can you please check if the code in this brach works for you?
worked on my machine.

https://github.com/LennartHennigs/Button2/tree/66-arduino-uno-r4-does-not-work-with-23

jay2216, 2024-02-18:
It does work. Thank you.

closed 2024-02-19 by LennartHennigs