LennartHennigs/Button2 #72: Problem with longpress behavior

juiplus, 2024-05-28:
I have tested the MultiHandler.ino example on an ESP8266. 
In principle it works, only the behavior with a long click does not work as expected. If a long click (handler setLongClickDetectedHandler() used) is executed after a single click or double click, the old state is always output first, e.g. click(1) or click(2). Only when a long click is executed again is it recognized as longclick correctly. 
 
Serial output:
Multi Handler Demo
click (1)
click (1) <-click loong, but click(1) will be recognized wrongly!?
longclick (1) <- that is correct

Please try to reproduce it, maybe there is a simple workaround.

LennartHennigs, 2024-05-29:
Hey Jui,
Thanks for reaching out. 
I will define check it out.

LennartHennigs, 2024-05-30:
Hey @juiplus,
you were right. Click count was also off. Updated the library.
Thanks again!

closed 2024-05-30 by LennartHennigs